elhep/MicroTCA-chassis #1: MTCA AMC box with RTM - issues

gkasprow, 2018-08-18:
_From @gkasprow on December 6, 2017 21:14_

- [ ] enlarge power entry cutout +1mm
- [ ] make AMC slot bigger to fix also EMC sealing. RTM side is OK
- [ ] PCB should be shifted to the front panel otherwise it's hart to unplug SFPs
- [ ] there is a conflict between backplane fixing surface and RTM connector
- [ ] conflict between  green rail guide and backplane fixing surface
- [ ] move green rail to front. RTM one is correct
- [ ] on the RTM side, the green rail conflicts with AMC panel EMC box
- [ ] add screw holes for wide panels used in ARTIQ project
- [ ] red rail conflict with chassis
- [ ] the enclosure length should be reduced by 2mm
- [ ] try to fit 60x60mm fans
- [ ] provide protection plate for SMPS
- [ ] add level shifter for JTAG
- [ ] add cutouts for twinax cable, just about the box floor
- [ ] add cutouts for fans, just about the box floor.

[image]
[image]
[image]

_Copied from original issue: sinara-hw/sinara#412_

pawelkasprowicz, 2018-12-07:
Are you sure  that AMC slot should  be increased? (I see that RTM is smaller one)
The details  of panels used in ARTIQ project needed
how the enclosure length should be reduced by 2mm? ( 1mm by both side?) 
"add cutouts for fans, just about the box floor"- not clear enough

gkasprow, 2018-12-07:
@pawelkasprowicz both AMC and RTM slots are identical. They use identical panels.
We use general purpose AMC panels, they are not ARTIQ specific.
After boards are fully plugged, the panel to panel distance is 2mm lower than box length. So you can cut it from either side. Anyway, the backplane is positioned using standoffs.
Cutouts for fans are needed for fan cables. Now they go via the main fan hole.

kaolpr, 2021-12-14:
What is the status of above issues? I can see that some of them are still not solved (e.g. AMC slot height).