elhep/MicroTCA-chassis #3: mechanical issues

gkasprow, 2018-12-04:
we need more place for ribbon cables and FFC jumpers
[image]

gkasprow, 2018-12-04:
Missig FFC cable cutout 
[image]

gkasprow, 2018-12-04:
Add venting cutouts
[image]

gkasprow, 2018-12-06:
elevated support wall by 1.1mm
[image]

kaolpr, 2021-12-14:
Is this solved in existing prototypes?

gkasprow, 2021-12-14:
I don't have a chassis to verify that, but it seems so.

kaolpr, 2021-12-14:
OK, I'll check it and close if solved.