elhep/MicroTCA-chassis #5: CPU selection

gkasprow, 2021-04-22:
@kaolpr We plan to replace the CPU anyway with something already used in the Sinara family (STM32). Or, maybe we should go for some single-board computer with Linux?

kaolpr, 2021-04-22:
SBC seems to be sort of appealing solution. The cost would not be much larger and at least at expected quantities it shouldn't be a problem. On the other hand, a CPU with MicroPython support available may be satisfactory. 

I imagine a simple web interface for managing the box would be a very pleasant feature. We don't need much here, a simple server with HTTP requests support + statically served JS web app for decent UX.

gkasprow, 2021-04-22:
Maybe the same approac we use in MCORD - STM32? We can run uPython and also RUST.

kaolpr, 2021-12-09:
I'd opt for [Omega2S](https://onion.io/omega2s/). It's SMT package, has integrated Ethernet PHY and all peripherals we need + WiFi w/uFL connector.

gkasprow, 2021-12-09:
we can, later on, upgrade to the 2s+ option with twice bigger memory.

gkasprow, 2021-12-09:
this particular SBC has the PCIe exposed. We can route it to SATA and in case we need, connect it to the backplane SATA connector which is connected to PORT2/3.

gkasprow, 2021-12-09:
of course, we would need to route the clock. but I'm not sure if that makes sense. The idea is to keep the project simple

kaolpr, 2021-12-14:
Don't think there is any reason to use PCIe.

Jest not to forget - when using Omega 2S, don't forget to add onboard reset button, so that bootloader update is possible.

closed 2022-10-28 by gkasprow