elhep/MicroTCA-chassis #6: USB / JTAG subsystem

kaolpr, 2021-12-09:
To make product more versatile, I'd like to suggest the following USB / JTAG subsystem:

[image]

- We would always have an entry-point with UART to the MCU.
- Additionally, if one wants to use USB JTAG, a Digilent programming module can be used ([link](https://digilent.com/shop/jtag-smt2-nc-surface-mount-programming-module/)) as an assembly option with FT2232, if no Xilinx Vivado compatibility is required. 
- Above path could be used in one of two ways: driven from external USB port or from the MCU (with SW mux driven by MCU). When driven from the MCU one could use USB over IP for remote debugging if, for some reason, remote debugging with XVC is not working or target is not Xilinx device.
- There is also option for plugging JTAG adapter directly. In such case BUFs are disabled.
- Onboard USB Host connector is foreseen for remote use of some debug adapter that is not in AMC chain (e.g. MMC). This connector would be on the main board, not exposed to the front panel.

Approximate costs:
- JTAG SMT2-NC: 49 USD
- FT2232: 4.25 USD
- USB-UART: 2.5 USD
- USB-HUB: 3.5 USD
- BUF: 4*0.8 USD = 3.2 USD
- SW: 0.44 USD

**Total:** 21.64 USD (FT2232) or 66.40 USD (SMT2-NC) + logistics + manufacturing

gkasprow, 2021-12-09:
can't we just use JTAG over SPI?

gkasprow, 2021-12-09:
can't single FT2232 be used as both UART and JTAG ?

gkasprow, 2021-12-09:
what about using a switch that connects FT2232 either to SBC to external micro USB (device). We could have two use cases:
1. UART + JTAG operated locally (USB plugged)
2. JTAG operated remotely (USB unplugged)
 I don't know if we need more scenarios

gkasprow, 2021-12-09:
actually, we can run Xilinx VCS over GPIO. 
https://github.com/kholia/xvcpi

gkasprow, 2021-12-09:
btw, FT2232 is unobtanium these days..

gkasprow, 2021-12-09:
I would keep it simple:
- use Cypress USB bridge as a serial terminal
- use JTAG header inside followed by the mux
- connect SBC IO to the mux and run XVCS
- use USB host connector inside
- when one wants to use a more sophisticated JTAG, one inserts it into the USB host and JTAG header

In this way, we have all scenarios as primarily mentioned.

gkasprow, 2021-12-09:
not all AMCs implement JTAG over AMC connectors. in this way we would connect the programmer between SBC host and native ZJTAG connector

kaolpr, 2021-12-14:
[image]

I appreciate simplicity. However, I'd opt for a minor additions:

* assembly variant for on-board FT2232 / JTAG-SMT2-NC, it's sort of appealing to have Xilinx debugging onboard without any additional elements,
* 2 onboard USB host connectors,
* 3 port USB 2.0 hub for serving onboard USB connectors and optional FT2232.

gkasprow, 2021-12-14:
Actually, we don't necessarily need a USB hub, we can use a switch instead. It's hard to buy any switch these days

gkasprow, 2021-12-14:
That's not that bad, some hubs are still available, i.e. USB2503-HZH

gkasprow, 2021-12-14:
yep, a few months ago it was much worse :)

gkasprow, 2021-12-14:
wow, even some UART bridges are again available ;D

gkasprow, 2022-10-28:
I used Atmega32U4 as a USB to master/slave I2C bridge due to FTDI issues

closed 2022-10-28 by gkasprow