elhep/MicroTCA-chassis #14: P2P, FP2, TCLKC, TCLKD

kaolpr, 2021-12-14:
Maybe we could add optional exposition of P2P, FP2 and TCLKC,D lines on the RTM side? 
I don't think this will be of very frequent use, maybe just make a mock PCB and prepare mechanics for the optional assembly?

gkasprow, 2021-12-14:
The P2P, FP2 lines can be assigned to the assembly pads and when required, the Twinax cables and thus the SFP cages could be connected to them instead.

kaolpr, 2021-12-14:
I was thinking of exposing them simultaneously with current set of connections.

gkasprow, 2021-12-14:
How? Maybe using optional SAS connectors?

kaolpr, 2021-12-15:
I was rather thinking of a PCB with SFP (in fact a section of main board) that would be placed beneath RTM slot, analogously to AMC side. Question would be - how to connect it to the backplane without compromising solution to #8. 

Can you give an example of what SAS connector you were thinking of?

As it is not expected to be very popular option, we could make some give-and-take, like use 3x USB-C vertical connector on the rear side of the backplane and connect SFP board with backplane through a cut-out in the separation wall.

gkasprow, 2021-12-15:
That makes sense. But that would be very rare use case. However it needs to be foreseen in the mechanical design. The issue arrives with soldering of the high speed coax cables. I mean internal SMT mini SAS connectors

gkasprow, 2021-12-15:
I meant [these](https://pl.farnell.com/multicomp/61v036s-321n0-b/mini-sas-connector-receptacle/dp/1735685?gclid=Cj0KCQiAweaNBhDEARIsAJ5hwbfPLpysWnvbyk6N8pVEgbM6l1IBuvwauDFi3CkxL0D5Q_F1TlaqSn8aAkJkEALw_wcB&mckv=scezk0iUJ_dc|pcrid|560370199601|plid||kword||match||slid||product|1735685|pgrid|130862485658|ptaid|pla-320184989749|&CMP=KNC-GPL-GEN-SHOPPING-Whoop-10-Nov-2021&gross_price=true) connectors.
+ [cables](https://www.amazon.com/Cable-Matters-Internal-Mini-Mini-SAS/dp/B01BW1ZCVW)

gkasprow, 2022-01-04:
Maybe we should switch to QSFP? With two QSFPs instead of 4 SFPs we can expose FP2 and FP1. In this way, we can easily route PCIe x4 to the PC...

gkasprow, 2022-01-04:
Actually, we can fit 3x QSFP+ instead of 4x SFP+

gkasprow, 2022-01-10:
@kaolpr what about leaving 2 SFPs for PORT0/PORT1 and adding 3x QSFP for FP1, FP2 and P2P ?
This of course means additional 4 Twinax ribbon jumpers, but we can always keep them DNP the same as the cages.

gkasprow, 2022-01-10:
The QSFP to 4x SFP cables are widely available.

kaolpr, 2022-01-14:
Greg, sorry for keeping it unanswered. I think having 3x QSFP is OK. They'll all go to the front panel? Maybe we could extend width of backplane (to go beneath the RTM connector) so that ribbons go more'n'less in straight manner?

gkasprow, 2022-01-14:
Yes, they fit on the front panel. longer ribbons mean more losses. We would have to switch to a much more expensive substrate. These ribbons don't cause issues while bending.

kaolpr, 2022-01-14:
I know, but it was already hard to approach panel with those ribbons. However, maybe we could stack them? Instead of going wide, make them go one on another (a bit like ribbons in Kasli). It combined with redesigned separation wall cutout could make it acceptably easy to assembly.

gkasprow, 2022-01-14:
Yes, the new boards will have stacked ribbons since we are adding a lot of connectors. Soldering would be far easier as well.

closed 2022-10-28 by gkasprow