elhep/MicroTCA-chassis #15: Software development

kaolpr, 2021-12-14:
Few points to include in a new software:

- System image build system based on Omega 2S OpenWRT distribution. I'd suggest this instead of distribution only an application so that we have more'n'less up to date base system and clear system update path.
- Use [ipmitool](https://github.com/ipmitool/ipmitool) for interacting with AMC modules.
- Implement temperature control subroutine.
- Define list of operations to be exposed in API.
- Create Python CLI application and library to interact with API.
- Create Web APP (JS) that would expose all the API functionality.
- Support for physical UI (#12).
- USB over IP remote configuration for two onboard USB Host ports.
- JTAG chain configuration.
- Xilinx Virtual Cable server implementation (see https://github.com/kholia/xvcpi).

gkasprow, 2021-12-14:
I would add some support for ARTIQ

kaolpr, 2021-12-14:
What support do you have in mind? We can easily make it ARTIQ network device, if we'll have Python library for API. But honestly I can't see much use of it (as it would mean using it from within the experiment).

gkasprow, 2021-12-14:
Maybe remote status ? Automatic bit update ? I don't know how usefull would it be.

jordens, 2021-12-14:
Integration into MQTT like we have for Stabilizer, Booster, and Thermostat would be great. That then gives you super-easy monitoring, ingest into time series databases, automatic logging and visualization of e.g. enviro and status data, configuration management and discovery.
An ARTIQ-specific integration isn't very meaningful IMO. It just needs high quality software on the device, an excellent API and good integration into protocols like MQTT.