elhep/MicroTCA-chassis #7: Front panel connectors

kaolpr, 2021-12-14:
Suggested set of front panel connectors:

- management Ethernet,
- 6x SFP (ports 0,1, 4-7),
- 2x RJ45 for MLVDS,
- [new] management WiFi SMA (SMA-u.FL cable),
- [new] TCLKA SMA (SMA-u.FL from the mainboard),
- [new] TCLKB SMA (SMA-u.FL from the mainboard).

gkasprow, 2021-12-14:
what about a micro-USB UART?
One USB host for the programmer would be inside, do we need another one on the front panel? I don't see any use case for one

kaolpr, 2021-12-14:
> what about a micro-USB UART?

Yes, I forgot about it. Editing comment.

> One USB host for the programmer would be inside, do we need another one on the front panel? I don't see any use case for one

I don't think putting USB host on FP is of any use.

gkasprow, 2021-12-14:
So the conclusion is that we need:
- USB UART on the panel
- USB A host on the mainboard for the external programmer, but installed still inside of the enclosure
- USB host permanently routed to the embedded JTAG programmer.

closed 2022-10-28 by gkasprow