elhep/MicroTCA-chassis #8: Backplane connection and assembly / serviceability

kaolpr, 2021-12-14:
In a current design Twinax cables must be soldered *after* main board and backplane are more'n'less in place (cables must be pulled through a slot between AMC and RTM areas). Despite the fact that it is way more difficult operation (yet possible), it makes any service action require desoldering Twinax cables.

I can see 3 solutions:

1. Extract SFP part to a separate PCB and make it out of better dielectric to keep performance of the MGT differential lines. On backplane we could use another AMC connector. In that way we could avoid soldering, remove Twinax cables and make assembly way easier. However, the cost increase *may*  be significant (to be verified). On the other hand this could also eliminate need for LVDS ribbon (not a big deal, but still less element in BOM).
2. Keep SFP part RF4, however add another AMC connector to backplane and mini-PCB to connect Twinax to backplane. This would shorten Twinax cables and keep all boards with cheaper dielectric.
3. Modify mechanics in a way that would allow pulling soldered backplane through a slot in a separation wall. This may be difficult due to stiffness requirements for AMC connector support.

gkasprow, 2021-12-14:
#3 would be the easiest. we can always mount the AMC backplane in front of the chassis, instead of behind it.

kaolpr, 2021-12-14:
It does not change much. AMC connector requires some force, I'm afraid that if we won't have mechanical connection to the bottom of the chassis, it won't be stiff enough. However, we can also try increasing separation wall thickness, there is room for that.

gkasprow, 2021-12-14:
That would change a lot. We would simply route the cables on the other side of the wall. No structural integrity would be affected.

kaolpr, 2021-12-14:
My mistake. Indeed it would change a lot. And it looks like a robust solution.