elhep/MicroTCA-chassis #19: v1.1 issues

gkasprow, 2024-07-01:
wrong PN and pinout for buch, either solder AP62201Z6-7 or update pinout

ras14k, 2024-07-24:
MAX6639AEE+ (IC1) and MAX6639ATE (IC9) are both soldered to the PCB. One of them has to be marked as DNP.

ras14k, 2024-07-24:
There's another issue related to the temperature management. Fans mounted in the current version of the mtca box (MF50151VX-1000U-G99) don't want to spin when PWM duty cycle is lower than 100%. It looks like this fan doesn't support PWM in power-supply, datasheet says:
"DO NOT use power or ground PWM to control the fan speed..."
https://www.tme.eu/Document/4048e5182dbe096f4ec79d2bcf98489e/MF50151VX-G99-A.pdf

I also checked different type of fans (a bigger one from Booster) and everything works fine. So we have to choose a different model of fans for the mtca box.

ras14k, 2024-08-13:
> There's another issue related to the temperature management. Fans mounted in the current version of the mtca box (MF50151VX-1000U-G99) don't want to spin when PWM duty cycle is lower than 100%. It [...]

Fonsoning FSY51S12M has the same dimensions and similar power. I bought two of these and tested them, they work great.